opalenzuela/dominoOSE #40: Opendomo v2.2 y domino 1.1.5

Juan2005, 2015-06-09:
Hola a todos, he instalado el domino 1.1.5 y anda bien tengo la placa arduino funcionando sin problemas.
Ahora quiero verlo con opendomo v2.2 pero cuando quiero agregar el controlador DominoOSE me muestra la leyenda "Cannot connect to the specified device".
En la configuración del URL para el controlador debo escribir solo la ip??? es decir en este caso 192.168.1.120??? o tiene otro tipo de sintaxis.
Para más datos:
Configuré una ip fija al opendomo y a la placa arduino si realizo el ping no tengo inconvenientes con ninguno.
Baje el plugin automatismo y el oddevel.
Gracias por la ayuda que me puedan brindar.
Saludos.

opalenzuela, 2015-06-10:
Debería ser la URL completa a la interfaz web, o sea "http://192.168.1.120/".

Juan2005, 2015-06-10:
Bien, coloq

Juan2005, 2015-06-10:
Gracias por la respuesta, coloque la dirección como me indicaste pero seguí con el mismo inconveniente, probé con 192.168.1.120 y no tuve suerte
Probé con ODControl2, la dirección 192.168.1.120 (sin la URL completa) y el dispositivo lo toma pero tengo fallas en la comunicación.
Para ser más claro en principio desde Opendomo me deja configurar los puertos del controlador me deja variar las opciones del pto, ejemplo si es on-off, el ícono (iluminación), el nombre de la variable.
Pero el problema es cuando veo el estado de los puertos siempre los observo en ON independientemente si los cambio desde la URL en forma directa o con el celular.
Si pude probar que cuando ambos están en ON(Opendomo y URL directa) y lo cambio a OFF desde opendomo si me toma el cambio(lo compruebo apagando 1 lámpara) pero cuando quiero volver a ON efectivamente no toma el cambio (no enciende la lámpara) debo cambiarlo desde la URL directa.
Y cuando refresco el estado de los puertos siempre los veo en ON.
Por otro lado probé con acciones para apagar la lámpara y no lo puedo hacer, tengo una leyenda que me indica ejecutando acción pero no hay cambios.
Mi gran duda es si esta situación pasa porque no está bien configurado desde el principio o es otro tipo de problema.
Muchas gracias por la respuesta q me puedas brindar.

Juan2005, 2015-06-10:
Otro dato más solo veo las salidas do006 a 009, las entradas no las veo, si en la URL directa.

opalenzuela, 2015-06-11:
Qué salida obtienes si accedes a http://192.168.1.120/lst  desde el navegador?

Juan2005, 2015-06-11:
Para ser más gráfico

[image]

Adjunto también la captura de los ports si sirve de dato

[image]

Gracias por la respuesta.

opalenzuela, 2015-06-12:
Ok, parece correcto. Podrías pasarme la salida de http://192.168.1.120/lsc ??

Juan2005, 2015-06-12:
[image]

Gracias por la respuesta

opalenzuela, 2015-06-30:
Procuraré retomar las pruebas a finales de verano. Ahora no dispongo del hardware necesario.

jordijo, 2016-02-22:
Alguna novedad? alguien ha conseguido agregar Domino a Opendomo 2.2